USACE/instrumentation #4: Add Instrument Error

lucasb16-exe, 2020-12-14:
Throws exception and displays white screen on attempting to add instrument.

Associated code according to debug console is highlighted. Find screenshot of error message in debug console attached.

[image]

Input for add instrument fields throwing error -
Project: Upper Appleton Locks and Dam
Name: UD-B-1
Type: Piezometer
Status: Abandoned
Status current as of: 12/14/2020
Station: *
Offset: 
Use Projection: WGS 84 (DD)
Longitude: -88.411711
Latitude: 44.253386
[image]

Error just started occurring today, any input would be appreciated, thanks!

KevinJJackson, 2020-12-14:
@lucasb16-exe - Hey Lucas, when was the last time you tried to add an instrument prior to this? I can take a look at some of the recent UI changes to see if something recently added caused this to go down or if you hit a weird snag that needs cleaned up.

lucasb16-exe, 2020-12-14:
@KevinJJackson

Hi Kevin, I had successfully added 4 instruments on 12/7/2020 to the Chicago Lock project. If there's anything you'd like me to try for troubleshooting, let me know. I'm looking at some of the changes from the upload-cleanup enhancement merge to see if there'd be some issue caused there.

KevinJJackson, 2020-12-14:
At first glance, I'd assume this line:
https://github.com/USACE/instrumentation-ui/blob/develop/src/app-components/domain-select.js#L20

Think that just needs to be set to:
```onChange={onChange}```
and it'd probably work as expected.

Let me see if I can test this locally first

KevinJJackson, 2020-12-14:
Yep, verified that was the culprit. I'll merge a hotfix for that line of code so that gets deployed ASAP

KevinJJackson, 2020-12-14:
@lucasb16-exe This should be deployed on dev in a few minutes or so. if you could try to reproduce again once it's deployed and report your findings here that'd be great!

lucasb16-exe, 2020-12-14:
Instruments successfully adding. Closing issue, thanks @KevinJJackson !

closed 2020-12-14 by lucasb16-exe