USACE/instrumentation #8: New Images for LRC-Chicago District Project Cards

lucasb16-exe, 2020-12-14:
Upload new images for projects...
- Thomas J. O'brien Lock and Dam
- Lower Appleton Locks and Dam
- Upper Appleton Dam
- Cedars Lock and Dam
- Chicago River and Harbor Controlling Works
- DePere Lock and Dam
- Chicago Sanitary and Shipping Canal Fish Barrier
- Indiana Harbor 
- J Edward Roush Dam
- J Edward Roush Dam - Markle Levee
- Kaukauna Locks and Dam
- Little Kaukauna Lock and Dam
- Little Chute Locks and Dam
- Lockport Lock and Controlling Works
- Menasha Lock
- Mississinewa Dam
- Rapid Croche Lock and Dam
- Salamonie Dam

Find named images in the attached file
[ChicagoDistrictProjectPictures.zip](https://github.com/USACE/instrumentation-ui/files/5638309/ChicagoDistrictProjectPictures.zip)

brettpalmberg, 2020-12-29:
Related to #9 .  Working on this now.

lucasb16-exe, 2021-01-05:
Find renamed images attached
[ChicagoDistrictProjectPictures (2).zip](https://github.com/USACE/instrumentation/files/5771914/ChicagoDistrictProjectPictures.2.zip)

lucasb16-exe, 2021-01-26:
Images updated for LRC Chicago, closing issue

closed 2021-01-26 by lucasb16-exe