USACE/instrumentation #18: Crash Upon Accessing/Creating Collection Group

lucasb16-exe, 2021-01-12:
[image]
Application crashes to white screen when attempting to create or access a collection group. See attached screenshot with error.

RangeError: Invalid time value

Not sure where a date object is getting passed containing an invalid date when just accessing a collection group

If any other information is needed let me know, any input would be appreciated
-Lucas

jeffsuperglide, 2021-01-19:
Looked at "Indiana Harbor" with @lucasb16-exe on this issue and it was reproducible.  I then tested `Collection Groups` for `Buffalo District Streamgages`.  Successfully added a group and timeseries to that group.  First difference between the two tested is Buffalo has data in timeseries and Indiana does not.  Further investigation into `Collection Groups` for projects without data.

lucasb16-exe, 2021-02-04:
Found source of error in [collectiongroup-timeseries-list.js](https://github.com/USACE/instrumentation-ui/blob/develop/src/app-pages/collectiongroup/collectiongroup-timeseries-list.js) line 52.

Looks like `item.latest_time` is passed into `parseISO()` without a check to see if it exists, if a timeseries selected for a collection group doesn't have any data, null value will be passed in causing crash.

Signing off for today, will fix via ternary tomorrow morning and pull.

lucasb16-exe, 2021-02-11:
Potential fix - line 52 -> `{item.latest_time != null ? formatDistanceToNow(parseISO(item.latest_time)) + ' ago' : 'No Data Found'}`

closed 2021-02-16 by lucasb16-exe