USACE/instrumentation #26: Sort timeseries data by time (newest at top)

adamscarberry, 2021-01-20:
Timeseries tab on instrument page

KevinJJackson, 2021-01-29:
@adamscarberry - This functionality has been added to the UI. The user now has the ability to sort the column by selecting the column header.

Added as part of: https://github.com/USACE/instrumentation-ui/pull/51/files#diff-1fe55ee43d67bda7b9197caf44665511795ca62b7815369f81c3cd8871cff77f

If this wasn't the expected result or if you need something else related to this, I am happy to take a look into it.

adamscarberry, 2021-01-29:
@KevinJJackson That looks great!.  Thank you.  There are some cases were default sorting is not ordered (which can be fixed by manually clicking the time header twice to sort correctly.  I'm assuming the default order coming from the API may be doing this?

Example: https://apps.rsgis.dev/instrumentation/flood-sensor-program/instruments/usace00411
Stage Param
Values on Jan 28th

[image]

KevinJJackson, 2021-01-29:
Correct, default ordering will always be as-is from the API unless otherwise stated. We could potentially pre-sort this data on the UI if you think there is value there? or if it is quicker to do that sort in SQL, it may be worthwhile to do it there instead.

adamscarberry, 2021-01-29:
Agreed.  The sorting should be addressed in the API SQL.  Thanks!

closed 2021-01-29 by adamscarberry