USACE/instrumentation #34: Wrong Page Numbers on Instrument Browser

lucasb16-exe, 2021-01-28:
[image]

Issue appears limited to projects with greater than 3 pages worth of instruments, repeats some page numbers and does not display others, will update issue with further investigation

KevinJJackson, 2021-01-28:
Can you provide browser and OS that you are using?

I'm running:
MacOS 10.15.7
Firefox 84.0.2

and I'm not seeing this issue, looking at Herbert Hoover Dike locally:
[image]

KevinJJackson, 2021-01-28:
Was able to partially reproduce this by changing the items per pages dropdown value, however:
[image]

lucasb16-exe, 2021-01-28:
Issue was reported by onsite park ranger at Salamonie, I'm also seeing this on production running 
with 10 instruments per page. I'll test locally and report back as well.

Google Chrome Version 88.0.4324.96 - 64 bit
Windows 10 Enterprise Version 18363.1139

KevinJJackson, 2021-01-28:
Think I found the cause, but if you could run this locally as well just to make sure. Looks like a type when creating the inner pages of the pagination component:
https://github.com/USACE/instrumentation-ui/blob/develop/src/app-components/pager.js#L53

instead of `i+i`, it should be `i+1`

lucasb16-exe, 2021-01-28:
Closing with merge, thanks Kevin

closed 2021-01-28 by lucasb16-exe