USACE/instrumentation #44: User Interface; Instrument Group, Add Existing Instrument, Modal Picklist Enhacement

adamscarberry, 2021-02-06:
I just added 40 existing instruments for the AWARE Pilot project and it's pretty tedious adding more than a few at a time.

Current modal allow a single select and disappears after click save changes (should that be "Add" ?)

Consider a multiple selection list where a user could use checkbox beside each selection.

Something like the concept below (no need for any type of hierarchy):

[image]

Bonus points for a filter search box to help reduce choices before selection :-)

KevinJJackson, 2021-02-10:
Hey @adamscarberry, I've been working on a multi-select component for the new batch plotting feature, does this seem like what you are looking for here as well:

[image]

I can spend some time making this filterable with input as well.

Which page/modal are you looking to see this added to?

adamscarberry, 2021-02-10:
Yes, that looks perfect.  

Example instrument group page: https://apps.rsgis.dev/instrumentation/flood-sensor-program/groups/nwp

[image]

I think it's in app-pages/instrument-picker.js, but please double check that's correct.

Thanks @KevinJJackson !

KevinJJackson, 2021-02-10:
Cool, I'll definitely take a look into this as soon as we get the batch-plotting code merged in (hopefully really soon with that!)

KevinJJackson, 2021-02-24:
@adamscarberry I've opened a PR to get this implemented. If you have no quarrels with the way this looks, I'll go ahead a merge. If there is anything else that you'd like added or changed once you've seen the implementation first hand, please let me know and I'll see what I can do.

https://github.com/USACE/instrumentation-ui/pull/66

[image]

KevinJJackson, 2021-02-24:
@adamscarberry Merged - As requested above, if you could provide feedback and if everything looks good, close at your discretion. Thanks!

closed 2021-03-01 by KevinJJackson