USACE/instrumentation #84: Explorer - Can't resize after landscape/portrait toggle (Shift +V)

brettpalmberg, 2021-03-15:
Steps to reproduce:

From home screen, select any project.  This will navigate to the `explorer`.

Toggle Portrait --> Landscape --> Portrait by typing `shift + v` then `shift +v`.

The panel splitscreen adjustment is locked.

KevinJJackson, 2021-03-22:
This is an underlying issue with React PanelGroup. I have opened an issue in that repo for further investigation and can be tracked here: https://github.com/DanFessler/react-panelgroup/issues/44

I have a "fix" locally that forces a re-render of the entire component tree underneath the PanelGroup, but this is very inefficient and has noticeable time delays with multiple data points on the Map. I suggest waiting for a fix in the component library and not moving forward with my temporary solution.

dennisgsmith, 2023-08-18:
Tested, now working as expected

closed 2023-08-18 by dennisgsmith