USACE/instrumentation #95: Deploy Develop and Stable Versions of Application

brettpalmberg, 2021-04-01:
[ ] UI Branch `stable` --> Bucket: `cwbi-apps-stable`
[ ] UI Branch `develop` --> Bucket `cwbi-apps-develop`
[x] Database `develop`
[ ] Database `stable`
[ ] API `develop`
[ ] API `stable`
[ ] Data Acquisition `develop`
[ ] Data Acquisition `stable`

brettpalmberg, 2021-04-01:
Related to deployment approach documented in #90

brettpalmberg, 2021-04-22:
Hello @KevinJJackson .  @adamscarberry and @brettpalmberg are working on migration to `develop` and `stable` instances.  Can you jump-in and help us on the items under `User Interface`?

brettpalmberg, 2022-12-14:
Removing myself from assignment; possible that this issue can be closed, as it appears to be substantially complete without additional comments for the past 2 years. Just doing some issue cleanup related to items in my "assigned" inbox.

KevinJJackson, 2022-12-16:
@brettpalmberg -  Yeah, taking a look into the backend as it seemed like we came across a few tables on a public DB that were causing some weird things to happen. @dennisgsmith - Feel free to close if you feel confident that the deployed DB is good where it is.

dennisgsmith, 2023-08-18:
All data has been migrated to `midas` schema from `public` and old tables removed.

dennisgsmith, 2023-08-18:
All data has been migrated to midas schema from public and old tables removed.

closed 2023-08-18 by dennisgsmith