USACE/instrumentation #97: Remove any In-Progress Features from Stable Version of User Interface

brettpalmberg, 2021-04-01:
@KevinJJackson  - can you help create a version of the user interface with any incomplete features removed?  Please feel free to remove anything else you see that's not quite ready for `stable` or is not implemented end-to-end yet.  List to include at least the removal of the following UI elements.  

* Landing Page `District/Division` Selector
* Instrument Manager Alerts Tab
* Instrument Manager Formula Editor Tab
* Instrument Manager Chart Tab

This is what will be deployed to `stable`

KevinJJackson, 2021-04-02:
Yep, I'll start with the above list and see if there is anything else that can be removed

KevinJJackson, 2021-04-02:
@brettpalmberg Created `stable` branch with the above features as well as the Cross Section portion on the Batch Plotting screen removed using some feature flags.

https://github.com/USACE/instrumentation-ui/tree/stable

Let me know if there is anything else that can be removed.

Current list of feature flags:
```
REACT_APP_DISTRICT_SELECTOR
REACT_APP_ALERT_EDITOR
REACT_APP_FORMULA_EDITOR
REACT_APP_INSTRUMENT_CHART
REACT_APP_CROSS_SECTION
```

Since these are all `undefined`, they evaluate to `false` even if they are not in the build command yet. If we ever wanted to turn these back on without a code change, we'd just need to set these values to `true` in the build command.

KevinJJackson, 2021-04-13:
Marking this as closed, feature flags can now be tracked here: https://github.com/USACE/instrumentation-ui/blob/develop/.github/workflows/development.yml

closed 2021-04-13 by KevinJJackson