USACE/instrumentation #102: Map Interactions (Pan/Select/etc.) Not Working

brettpalmberg, 2021-04-08:
Map interactions are not working in application where it is newly deployed at:

- midas.rsgis.dev
- develop-midas.rsgis.dev

brettpalmberg, 2021-04-08:
@KevinJJackson - we fixed this one yesterday.  I just created an issue for tracking.  Can you add any details we want to document, reference the PR, and close this issue at will?

KevinJJackson, 2021-04-12:
This issue was caused by an extraneous import of a css file from a package that we thought wasn't being used. By moving this import to our application, the need for the corpsmaps packages are no longer necessary.

https://github.com/USACE/instrumentation-ui/pull/81 - removed the packages and imported the css file required for mapping functionality to work as normal.

closed 2021-04-12 by KevinJJackson