USACE/instrumentation #114: [BUG REPORT]: Uploader not parsing lat/lon from csv file

adamscarberry, 2021-05-06:
## Describe the bug

Unable to upload test file below.  Lat/lon do not populate the parsed tab.

[Instruments_test_upload.csv](https://github.com/USACE/instrumentation/files/6435752/Instruments_test_upload.csv)

## Steps To Reproduce
1. Select project from homepage
2. click uploader
3. select csv file, upload
4. select appropriate fields on left to map

## Expected behavior
Lat and lon should parse, then allow successful upload.

## Actual behavior
lat/lon will not parse.  upload will fail

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Linux
 - Browser Firefox
 - Version 88

KevinJJackson, 2021-05-07:
Yeah this is a strange one, it recognizes that you've selected a value and says it doesn't have an error b/c of it but isn't saving the value to the parsed data array on the store side... will have to take a dive into the uploader code a bit more. will update when i have more info

KevinJJackson, 2021-05-07:
It actually looks like the error you were getting with the upload was actually due to the offset being a string and not a number, so I've updated that and also made it so you can see the value in long/lat columns of the parsed table