USACE/instrumentation #119: [BUG REPORT]: Constant Values Not Saving Past UI

lucasb16-exe, 2021-06-02:
## Describe the bug
Constants can be declared for an instrument, but added values to the constant are not saving or usable in the formula editor. Values can be added to the declared constant, but after switching tabs or menus, the values disappear.

## Steps To Reproduce
1. Open any instrument in MIDAS
2. Navigate to the 'Constants' tab
3. Declare a constant
4. Add a value to the constant
5. Change tabs (try to use the formula editor with the new constant)
6. Go back to the 'Constants' tab
7. The value you added will be gone

## Expected behavior
Utilize constant values for the formula editor

## Actual behavior
Constant values appear to not save past the UI

## Screenshots
[image]
[image]
[image]
[image]

## Desktop:
 - OS: Windows 10
 - Browser: Chrome
 - Version 90.0.4430.212

## Additional context
Investigating with Jeff Gregory

jeffsuperglide, 2021-06-03:
I've confirmed this in stable and develop.  The added constant shows in the formulas tab but doesn't not result in a working formula.

KevinJJackson, 2021-06-10:
After a bit of tweaking, I was able to get the values to successfully update as well as new values persisting. I was also able to get constant deleting to remove right away, instead of needing a refresh. However, there is a bit of a glitch on the API side of things where updating the date-time causes row duplication. This along with the ability to delete particular entries will be added when @dliang864 is able to make the changes in the api early next week.

I'll keep this issue updated as more info becomes available.

KevinJJackson, 2021-06-24:
@lucasb16-exe - This should be working now in dev. Sorry for the delay, was hoping we get some more fixes around constants as part of this work.

As of now, creating a new constant will persist. You can also go back and update the **value** of any of them. However, we are still working out some details on how we can go about **updating the time** and **deleting a value** altogether.

lucasb16-exe, 2021-06-24:
@KevinJJackson Sounds great, thanks Kevin. Any expectation yet for when we could see it in stable?