USACE/instrumentation #120: [BUG REPORT]: Water Level Instrument Timeseries not adding to Collection Groups

lucasb16-exe, 2021-06-24:
## Describe the bug
Timeseries from 'Water Level' instrument types can not be added to collection groups from UI. Will continue to investigate to find other instrument types or features 'Water Level' instrument timeseries could be excluded from.

## Steps To Reproduce
1. Create an instrument and select 'Water Level' as instrument type
2. Add a 'Depth' timeseries to the instrument
3. Create a collection group in the same project as the 'Water Level' instrument type
4. Attempt to add the Water Level instrument's Depth timeseries to the created collection group

## Expected behavior
Water Level depth timeseries successfully added to collection group

## Actual behavior
Water Level depth timeseries does not add, API receives POST request with code 200 but either does not add requested timeseries to db or UI does not represent added timeseries

## Screenshots
[image]
[image]
[image]
[image]

## Desktop (please complete the following information):
- OS: Windows 10
- Browser: Chrome
- Version 90.0.4430.212

## Additional context
Will investigate further to see if any other instrument types have this problem

lucasb16-exe, 2021-06-24:
API successfully returns all timeseries in collection group, but UI selector 'selectCollectionGroupDetailByRoute' only returns timeseries with data

KevinJJackson, 2021-06-24:
@brettpalmberg | @adamscarberry | @dliang864 - When someone gets a chance, can you take a look into the collection group details here https://github.com/USACE/instrumentation-api/blob/stable/models/collection_group.go#L59

Looks to me like it can't return the required timeseries info unless there has been a measurement with the same timeseries_id. The api should be able to return the timeseries data in the response as normal and leave the `latest_value` and `latest_time` as null.

lucasb16-exe, 2021-10-13:
In https://github.com/USACE/instrumentation-api/blob/stable/models/collection_group.go#L59 - 
 

```
// GetCollectionGroupDetails returns details for a single CollectionGroup
func GetCollectionGroupDetails(db *sqlx.DB, projectID *uuid.UUID, collectionGroupID *uuid.UUID) (*CollectionGroupDetails, error) {
	var d CollectionGroupDetails
	// Query 1
	if err := db.Get(&d, listCollectionGroupsSQL+" WHERE project_id=$1 AND id=$2", projectID, collectionGroupID); err != nil {
		return nil, err
	}
	// Query 2
	d.Timeseries = make([]cgdTsItem, 0)
	if err := db.Select(
		&d.Timeseries,
		`SELECT t.*, tm.time as latest_time, tm.value as latest_value 
		FROM collection_group_timeseries cgt 
		INNER JOIN collection_group cg on cg.id = cgt.collection_group_id 
		INNER JOIN v_timeseries t on t.id = cgt.timeseries_id 
		INNER JOIN timeseries_measurement tm on tm.timeseries_id = t.id and tm.time = (
			select time from timeseries_measurement 
			where timeseries_id = t.id 
			order by time desc limit 1) 
		WHERE cgt.collection_group_id = $2 and t.project_id = $1
		 `, projectID, collectionGroupID,
	); err != nil {
		return nil, err
	} else if len(d.Timeseries) == 0 { //  if query isn't an error but doesn't return anything, execute second query disregarding latest measurement
		if err := db.Select(
			&d.Timeseries,
			`SELECT t.*
			FROM collection_group_timeseries cgt 
			INNER JOIN collection_group cg on cg.id = cgt.collection_group_id 
			INNER JOIN v_timeseries t on t.id = cgt.timeseries_id 
			WHERE cgt.collection_group_id = $2 and t.project_id = $1
			 `, projectID, collectionGroupID,
		); err != nil {
			return nil, err
		}
	}

	return &d, nil
}
```

lucasb16-exe, 2022-03-10:
addressed in [USACE/instrumentation-api/pull/128](https://github.com/USACE/instrumentation-api/pull/128)

will resolve issue after review and merge

lucasb16-exe, 2022-03-10:
fix merged to dev, resolved issue, closing

closed 2022-03-10 by lucasb16-exe