USACE/instrumentation #143: [BUG REPORT]: QA/QC Tab permissions/Boxes hidden

ScottMassa, 2023-11-07:
## Admin User could not see "Submit New Evaluation" or "{Create New Alert Config" boxes on the QA/QC Tab

## Steps To Reproduce
Log in, Got to QA/QC tab, view screen

## Expected behavior
I am able to see user input boxes, the Admin user can not see them.

## Actual behavior
No boxes to input evaluation or set up the alerts

## Screenshots
If applicable, add screenshots to help explain your problem.

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser Microsoft Edge

## Additional context
Add any other context about the problem here.

KevinJJackson, 2023-11-09:
Hi Scott, thanks for bringing this to our attention.

This functionality seems to be hidden behind the same logic that hides the `#data-logger` and `#admin` tabs on the project page. This feature utilizes the profile.roles object, stored in the DB, to evaluate if users should have access or not. This is the standard use for allowing project admins access: `${project.slug.toUpperCase()}.ADMIN`.

In your use case, I _believe_ you are set up as an application admin and can view everything on the site and bypass the project role-check.

Can you verify with the user that they can see and access the above listed tabs?
If not, can you send @dennisgsmith the user's information and he can do a quick DB lookup to see if that user has the proper access. For example, a user with admin rights to the Blue Water Dam Example Project, would have a role like:
`['BLUE-WATER-DAM-EXAMPLE-PROJECT.ADMIN']`.
If there is still an issue beyond that, we can take a look into deeper nested logic that would take more time to resolve.

dennisgsmith, 2023-11-09:
@ScottMassa What was the specific project that the user was trying to access? Is this issue persistent throughout all projects? We need to validate that the assigned `<project>.ADMIN` role is present for the project in question.

ScottMassa, 2023-11-09:
It was user Sarah.Wiles ***@***.******@***.***> on the J Percy Priest QA/QC tab. I think it happens on other projects in her district (Wolf Creek, Center Hill, Old Hickory, etc.)

dennisgsmith, 2023-11-17:
Fixed with https://github.com/USACE/instrumentation-ui/pull/193

closed 2023-11-17 by dennisgsmith