USACE/instrumentation #144: [BUG REPORT]: User Permissions for Collection Groups

ScottMassa, 2023-11-16:
Non Admin and Non User was able to enter data into a collection group

Was notified that a user from Louisville District was asked to enter data into a collection group on a project they were not a member (user or admin). Data was populated into the database. 

## Expected behavior
Would not have access to write to the database. 

## Actual behavior
User was able to submit a timeseries record that got populated into the database. 

## Screenshots
If applicable, add screenshots to help explain your problem.

## Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Browser [e.g. Microsoft Edge]
 - Version [e.g. 90.0.818.46]

## Additional context
I can request user name and project if necessary.

dennisgsmith, 2023-11-16:
@ScottMassa Thanks for reporting this issue. This falls in line with our discussion in the Sprint Planning meeting about a month ago and is due to some historical reasons in the MIDAS API.

## Summary
For a quick fix, @KevinJJackson can hide this button on the UI from users. Are you able to send an exact screenshot of the exact button/page that should be hidden?

## Detail
MIDAS has only ever done this check on the client (UI) and the buttons are simply hidden from users based on their project role. It has never been not enforced by the server -- any user with a MIDAS account on the network can make requests to the `private` group, which is not ideal.

There are 4 types of external request groups for the API (all restricted by USACE network):
1. `public` (does not need to be registered user)
2. `private` (must be registered user, pass key auth or CAC, but not a member or admin of any project)
3. `cacOnly` (must be registered user, pass CAC, but not a member or admin of any project)
4. `applicationAdmin` (superuser, only select number of users, manually granted by developer)

To correctly implement project role authentication, we want to add project-level restrictions to specific endpoints that use the private and/or cacOnly groups. It actually looks like the functionality is partially implemented in the API codebase but was never completed. In order to complete this, **it would require us to have well-defined permissions for each endpoint (for example, which endpoints should be project member vs project admin, etc.).**

If you could supply me with a list of endpoints and their respective project roles via email, I can finish the implementation, based on the endpoints defined in the swagger doc: https://develop-midas-api.rsgis.dev/swagger/index.html. Alternatively, depending on your availability, we can set up a meeting to discuss this synchronously.

dennisgsmith, 2023-11-17:
@ScottMassa A fix was pushed to the UI yesterday evening to address this ticket -- please let us know if this resolves the issue you. mentioned. Additionally, I'd be happy to discuss any details or questions you have about my above comment. Thanks!

ScottMassa, 2023-11-17:
I’ve asked the Louisville District to test, will let you know when I hear back.

Scott

closed 2024-01-30 by dennisgsmith