USACE/instrumentation #150: [BUG REPORT]: UI will not save updated Station and Offset

ScottMassa, 2023-12-19:
## Describe the bug
User tries to modify the station and offset, updated values are seen, but not saved.

## Steps To Reproduce
Edit instrument, modify station and offset, click save, the UI shows the new values, but it is not saved. 

## Expected behavior
new values are expected to be saved.

## Actual behavior
Not saved, API error. ( see screenshot below)

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser Microsoft EDGE
 - Version 120.0.2210.77

## Additional context
User also mentioned similar when adding constants or timeseries to an instrument.

ScottMassa, 2023-12-19:
Also similar behavior when adding instruments to plot groups. i try to add an instrument and after saving it doesn't appear in the plot groups. no put error in MS Edge inspector this time.

dennisgsmith, 2023-12-26:
@KevinJJackson The PUT payload send to `/projects/:project_id/instruments/:instrument_id` should include `type_id` field, instead it's null.

I'll look into the instrument group issue separately and report back.

dennisgsmith, 2023-12-28:
> Also similar behavior when adding instruments to plot groups. i try to add an instrument and after saving it doesn't appear in the plot groups. no put error in MS Edge inspector this time.

This was a separate bug, fixed in https://github.com/USACE/instrumentation-api/pull/203

dennisgsmith, 2024-01-31:
Fixed deployed in https://github.com/USACE/instrumentation-api/pull/210

closed 2024-01-31 by dennisgsmith