USACE/instrumentation #163: [BUG REPORT]: Telemetry Data Not Importing

ScottMassa, 2024-02-08:
## Describe the bug
Telemetry data not being populated to database, but is posted to the API

## Expected behavior
Data is saved in database based on mapping table

## Actual behavior
Data not populated, want to confirm functionality of mapping table, is it on "unnamed table" or should it be per table?

## Screenshots
[image] No Field mapping table

## Desktop (please complete the following information):
 - Browser Microsoft Edge

## Additional context
Add any other context about the problem here.

dennisgsmith, 2024-02-08:
Cross-posting information from email to this ticket for posterity:

> There was an edge case in the Telemetry API code preventing some existing equivalency mapping tables from migrating. I just pushed a fix to production for this.
>  
> The expected workflow is:
> Existing table names take on the name of their `table_name` field automatically.
> If that `table_name` already exists for that datalogger, it will rote the telemetry to that table instead.
>  
> The issue was that the original datatlogger payloads did not account for the table name, so all the table names that were empty should have been renamed to what was in the incoming payload. Instead, an additional datalogger table was created on top of the existing default datalogger table (before the “multiple equivalency table” changes).
>  
> For new datalogger tables, this worked fine, as the data tables would get created when the telemetry data was POSTed and the equivalency tables could automatically be created from the payload.
>  
> Please note – if there were any new equivalency tables created that should have already had an equivalency table from before this feature, they will need to be deleted for this fix to take effect.

dennisgsmith, 2024-02-12:
Closing, this should be fixed with https://github.com/USACE/instrumentation-api/pull/212

If you are still experiencing related issues, please feel free to reopen this ticket.

closed 2024-02-12 by dennisgsmith