USACE/instrumentation #164: [BUG REPORT]: Formula Editor

ScottMassa, 2024-02-20:
## Describe the bug
The Formula Editor now shows instrument timeseries name as requested instead of slug. It also shows all timeseries per project. Problem is its unclear which timesieries is which. The name "riser" is shown multiple times for example. 

## Steps To Reproduce
1. Go to instrument screen
2. Click on formula editor
3. Scroll down to see multiple entries

## Expected behavior
Available parameters would be instrument and timeseries.  Maybe the instrument shown on the page would be listed first, then other options (maybe searchable) would be below. 

## Actual behavior
Large list of parameters, not sure which is which.

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser:  Microsoft Edge
 - Version 121.0.2277.128

dennisgsmith, 2024-02-21:
@ScottMassa Thank you for filing a bug report. @KevinJJackson and I are assessing this issue and working towards a fix.

ScottMassa, 2024-02-21:
> Hi @ScottMassa<Blockedhttps://github.com/ScottMassa> -- after discussing this with @KevinJJackson<Blockedhttps://github.com/KevinJJackson>, it seems like the most straightforward solution would be to simply use the entire variable in the lefthand list, in order to more easily differentiate which instrument the parameter belongs to, while still allowing cross-instrument formulas to exist. The proposed change would change the list on the left as follows:
>
> battery -> test-instrument-1.battery
>
> battery-2 -> test-instrument-2.battery-2
>
> battery-3 -> test-instrument-3.battery-3
>
> This will reflect exactly as they appear in the formula editor after double-clicking the parameter. Feel free to provide any feedback you have on this solution.

I think this looks good. Is there any way show only the instrument, then click a button to expand to all the others? Some projects may show several hundred timeseries to choose from.

Scott

KevinJJackson, 2024-09-23:
@ScottMassa - Updating this to have timeseries sorted underneath their respective instrument. Here is the proposed UI:

[image]

Does this look acceptable for the use case described above? We can also add a search bar within each instrument or a general search bar that could have all the timeseries (again grouped by instrument) outside the boxes, similar to what we added to the Plotting screen, where selecting an item from the dropdown would act the same as double clicking the timeseries as current:

[image]

KevinJJackson, 2024-09-23:
Went ahead with the global search:
[image]

ScottMassa, 2024-10-11:
Can we use "Names" instead of slug/variable in these?  It would be more friendly to the user and reflect changes if they were made, unlike the slugs.  Some slugs/variables are like the following and the user doesn't need to know we have over 2 C2s > 1673 depths to water.  c2-3.depth_to_water-1673

dennisgsmith, 2024-11-22:
Going to spin this out into it's own ticket since it's different from the original one submitted, which has been merged.

Now tracking variable rename request in https://github.com/USACE/instrumentation/issues/209