USACE/instrumentation #165: [BUG REPORT]: Mask Fields not working

ScottMassa, 2024-03-07:
## Describe the bug
The mask field does not store the current status when boxes are selected

## Steps To Reproduce
1. Go to instrument screen
2. Click on timesieries, click a value that needs to be masked, It does not retain the check mark.

## Expected behavior
When a box is selected the state is stored

## Actual behavior
When a box is selected the state is not stored

## Screenshots

## Desktop (please complete the following information):
 - OS: Win 10
 - Browser Microsoft Edge

## Additional context
Add any other context about the problem here.

KevinJJackson, 2024-03-08:
Thanks for the report, we are investigating this issue and will report back when a solution is found.

dennisgsmith, 2024-03-11:
Closed with [#215](https://github.com/USACE/instrumentation-ui/pull/215)

closed 2024-03-11 by dennisgsmith