USACE/instrumentation #169: [BUG REPORT]: Uploader

ScottMassa, 2024-04-11:
## Describe the bug
The uploader fails with "400 bad request" when data file has several thousand records

## Steps To Reproduce
1. Go to uploader
[Uploading RPool.csv…]()

2. Click on choose file, select RPool.csv
3. Import as TS measurement
4. See error

## Expected behavior
Returns "201 Created" and uploads successfully.

## Actual behavior
Returns 400 bad request but will upload when you reduce the file from 7500 rows to 1000 rows. 

## Screenshots
[image]

dennisgsmith, 2024-04-11:
@ScottMassa, thanks for reporting this.

- Is this a new issue (i.e. were you previously able to upload documents over 1000 rows using this tool)?
- When was the first time you encountered the issue?

ScottMassa, 2024-04-12:
We could previously upload larger data sets. I can’t remember exactly but probably 50k to 100k records . One of our users in Pittsburg asked me about the uploader being broken this morning. I gave it a try and got the 400 error, then tried it with 1000 records and it worked. I also checked to make sure there were no bad characters in the values.

@ScottMassa<Blockedhttps://github.com/ScottMassa>, thanks for reporting this.

dennisgsmith, 2024-04-12:
Thanks Scott. I'm looking into this now. If you could provide the response body sent back with the failed request (you can get this from the "Response" tab in the network section of the Devtools) it would be helpful for us to diagnose and resolve this issue. If you could also send me an email with the offending file attached, that would also be very helpful.

ScottMassa, 2024-04-12:
Info is below and  attached.,

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>400 Bad Request</title>
</head><body>
<h1>Bad Request</h1>
<p>Your browser sent a request that this server could not understand.<br />
</p>
</body></html>

Request URL:
https://midas.sec.usace.army.mil/api/projects/5b6f4f37-7755-4cf9-bd02-94f1e9bc5984/timeseries_measurements?dry_run=true
Request Method:
POST
Status Code:
400 Bad Request
Remote Address:
3.30.180.152:443
Referrer Policy:
strict-origin-when-cross-origin

dennisgsmith, 2024-04-12:
@ScottMassa We believe this issue lies int he configuration of the reverse proxy. I have submitted a ticket to CWBI for further investigation: [CWBI-21280](https://jira.il2.dso.mil/browse/CWBI-21280). Please see this ticket for further information.

dennisgsmith, 2024-05-15:
This this issue has been resolved in TEST and PROD by updating the WAF to allow larger payloads, see the liked ticket above for more details.

closed 2024-05-15 by dennisgsmith