USACE/instrumentation #171: [BUG REPORT]: Uploader tool

ScottMassa, 2024-05-08:
A CSV of instruments fails to upload when imported as instruments

## Steps To Reproduce
1. Go to Blue water dam, uploader tab
2. select file, 
Example CSV:
 Project,Name,Status,Type,Longitude,Latitude,ZReference,Station,Offset
My Sample Project Name 1,BWD-2-1,active,Drain,-80,30,,100,50
My Sample Project Name 1,BWD-3-1,active,Drain,-80.05,30.05,,100,50
My Sample Project Name 1,BWD-3-2,active,Drain,-80.1,30.1,,100,50
My Sample Project Name 1,BWD-3-3,active,Drain,-80.15,30.15,,100,50
My Sample Project Name 1,BWD-3-4,active,Drain,-80.2,30.2,,100,50
My Sample Project Name 1,BWD-3-5,active,Drain,-80.25,30.25,,100,50
My Sample Project Name 1,BWD-3-6,active,Drain,-80.3,30.3,,100,50
My Sample Project Name 1,BWD-4-1,active,Drain,-80.35,30.35,,100,50
My Sample Project Name 1,BWD-4-2,active,Drain,-80.4,30.4,,100,50
My Sample Project Name 1,BWD-4-3,active,Drain,-80.45,30.45,,100,50
My Sample Project Name 1,BWD-4-4,active,Drain,-80.5,30.5,,100,50
My Sample Project Name 1,BWD-4-5,active,Drain,-80.55,30.55,,100,50
My Sample Project Name 1,BWD-4-6,active,Drain,-80.6,30.6,,100,50
3, select import as "instrument"
[Inst_List.csv](https://github.com/USACE/instrumentation/files/15253277/Inst_List.csv)

4. select name, status, etc. 
5. Upload. 

## Expected behavior
Expect instruments to be created, get error that the name is already taken. 

## Actual behavior
No upload

## Screenshots
If applicable, add screenshots to help explain your problem.
[image]

## Desktop (please complete the following information):
Windows 10
Microsoft Edge 124.0.2478.80 (Official build) (64-bit)

dennisgsmith, 2024-05-09:
Thank you for submitting a bug report. We're looking into this issue now and will follow up shortly.

dennisgsmith, 2024-05-15:
This issue has been resolved and is currently being validated in CWBI-TEST before deployment to production. This ticket will be closed once the fix is rolled out to production.

dennisgsmith, 2024-08-16:
Closed with https://github.com/USACE/instrumentation-api/pull/231

closed 2024-08-16 by dennisgsmith