USACE/instrumentation #173: [BUG REPORT]: Unable to Login

ScottMassa, 2024-06-10:
## Describe the bug
Users cannot login to MIDAS

## Steps To Reproduce
1. Go to MIDAS
2. Click on Login
3. Nothing happens

I then inspected the site and saw no response from https://corpsmap-dev.sec.usace.army.mil/cwbi/goauth/token/(my uuid)

## Screenshots
[image]

[image]

## Desktop (please complete the following information):
 - OS: Windows10
 - Browser : Microsoft Edge

## Additional context
Add any other context about the problem here.

dennisgsmith, 2024-06-10:
@ScottMassa Thanks for reporting this issue. At a glance it seems that is is an issue with the authentication server rather than MIDAS calling it. I will investigate to confirm this and report this to CWBI as needed.

dennisgsmith, 2024-06-10:
@ScottMassa I have tested and confirmed this behavior on my end. I found that there is currently an open issue in CWBI tracking this: https://jira.il2.dso.mil/browse/CWBI-21855
I will push toward getting this resolved and keep you updated as the ticket progresses.

Rockscientist, 2024-06-10:
Thanks!Sent from my iPhone

dennisgsmith, 2024-06-17:
This issue has been resolved, but has resulted in effort to migrate to CWBI Keycloak instance.

closed 2024-06-17 by dennisgsmith