USACE/instrumentation #175: [BUG REPORT]:

ScottMassa, 2024-07-22:
## Describe the bug
Uploader Tab inserts "comment" into blank Annotation fields when uploading comments. 
## Steps To Reproduce
1. Go to uploader
2. Click on timesieries data
3. Upload file that has some blanks in the annotation field,
4. view parsed. Upload, see "comments" in the annotation field.

## Expected behavior
Expect valid comments, and blanks where no comment exists.

## Actual behavior
inserted "Comment" in blank fields.

## Screenshots
If applicable, add screenshots to help explain your problem.
[image]
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser; Microsoft Edge | 126.0.2592.113

KevinJJackson, 2024-07-22:
@ScottMassa - Can you provide the file you were attempting to upload (or a partial of that file)? I believe this should be an easy fix, but would like to see it tested with your data.

KevinJJackson, 2024-07-22:
@ScottMassa - This is a pretty old bit of code I'm changing, so just want to make sure this change won't affect any expected workflows.

This code was explicitly written to always have a default value if a value was selected to be parsed. In this case you selected the 'Comment' column of the csv to be provided to the Annotation field, which is why the word 'Comment' was supplied to empty fields. If the column name in the csv was changed to 'Notes', the empty fields in the Annotation would have been replaced with 'Notes'.

I can change this functionality to only ever parse data that exists in the csv, though this could change how some people were populating large data sets with a default string. For example, if a user wanted all of the fields to be filled with the same string, they could set the column name in the csv to that string, and it would parse that through every row. With this change, the user would need to set the string on each row in the csv.

I just wanted to confirm this change with you before it gets made.

ScottMassa, 2024-07-23:
Lets change it to use the data that exists in the CSV. If a user wants to import a thousand records with the same comment, they can copy that into the CSV. The way it is now is, if you have a thousand rows and only want to add notes to 3 of them, we have 997 other values.

Thanks,
Scott

dennisgsmith, 2024-08-23:
Fixed with https://github.com/USACE/instrumentation-ui/releases/tag/v0.16.0

closed 2024-08-23 by dennisgsmith