USACE/instrumentation #178: [BUG REPORT]: Missing Mask, Validated, Annotation Fields

ScottMassa, 2024-08-15:
## Describe the bug
Missing Mask, Validated, Annotation Fields

## Steps To Reproduce
1. Go to Instrument screen, Timesieries tab

## Expected behavior
Be able to mask data

## Actual behavior
No mask field exists

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser: Microsoft Edge, 127.0.2651.98

KevinJJackson, 2024-08-15:
@ScottMassa - Thanks for reporting, this issue was already being tracked and will be fixed in the next release. I will tag the corresponding PR and close this ticket upon completion.

dennisgsmith, 2024-10-24:
Closing as completed.

closed 2024-10-24 by dennisgsmith