USACE/instrumentation #179: [BUG REPORT]: MIDAS Login with Login.gov

ScottMassa, 2024-08-16:
## Describe the bug
Remnants of old login system still in place

## Steps To Reproduce
Login from Gov Computer

## Expected behavior
Directed to Login.Gov after clicking on Login

## Actual behavior
ActivClient asks for PIN, Then Directed to create profile.

## Screenshots
[image]
[image]
After CTRL+R and navigating back to https://midas.sec.usace.army.mil/midas/
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser: Edge

dennisgsmith, 2024-08-16:
Hi Scott, we suspect there was a caching issue -- I have deployed a fix for this. Please let me know if this issue persists.

dennisgsmith, 2024-08-21:
Closing as fixed, please reopen if this issue persists.

closed 2024-08-21 by dennisgsmith