USACE/instrumentation #180: [BUG REPORT]:Edit instrument screen

ScottMassa, 2024-08-20:
## Describe the bug
Changes made to an instrument do not update the API

## Steps To Reproduce
go to an instrument, edit station and offset, click save.

## Expected behavior
API Post is successfull

## Actual behavior
API returns 400 Bad Request in inspector, also save button does not dismiss instrument screen

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser Microsoft Edge
 - Version 127.0.2651.105

dennisgsmith, 2024-08-20:
@ScottMassa Thanks for this report, we are currently looking into this issue and will update you on the status once resolved.

dennisgsmith, 2024-08-21:
Fixed in https://github.com/USACE/instrumentation-ui/pull/234

closed 2024-08-21 by dennisgsmith