USACE/instrumentation #181: [BUG REPORT]: Upload Timeseries data via Uploader

ScottMassa, 2024-08-22:
## Describe the bug
Uploader tool returns "API Error: undefined" when uploading a timeseries measurement

## Steps To Reproduce
1. Go to Uploader tab and import timeseries measurement
2. Click Upload.

## Expected behavior
Data is uploaded.

## Actual behavior
API error: Undefined, and 400 Bad Request

## Screenshots
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser: Microsoft Edge
 - Version 127?

ScottMassa, 2024-08-22:
Data is not update, also TS UUID is not populated in the Parsed Screen, See attached.
[image]

dennisgsmith, 2024-08-22:
@ScottMassa thanks for the bug report. I'm currently looking into this -- it seems there is an issue with the uploader not populating the timeseries_id field in the request payload:
[image]

I'll let you know once this issue is resolved.

dennisgsmith, 2024-08-22:
@ScottMassa Looking at the code, it seems that the parser expects a "timeseries_id" column in the uploaded csv. Is this the intended behavior? Or it the expectation to be able to select a single timeseries from the dropdown list within the uploader?

@KevinJJackson is currently out of office -- I'll attempt to resolve on my own but will make sure he's aware of this issue once he returns tomorrow.

ScottMassa, 2024-08-22:
We have always selected the timeseries ID from the drop down list, or it’s been prepopulated if we select upload from the timeseries on the instrument tab.  The next scope of work will have an equivalency table so we can upload multiple timeseries at once.
I think this feature broke since last Wednesday night, we just noticed it today.

Scott

KevinJJackson, 2024-08-23:
Fixed in latest release https://github.com/USACE/instrumentation-ui/releases/tag/v0.17.2

closed 2024-08-23 by KevinJJackson