USACE/instrumentation #182: Uploading (1a)

madisonschattenfield, 2024-08-23:
### **DESCRIPTION**
Batch upload of data – provide a user-friendly method of upload of timeseries data with multiple instruments. Data will include name, date/time (non-ISO8601, MM-DD-YYYY HH:MM) and values in multiple columns. The user should map the name to instrument and timeseries and save the mapping for future uploads. Convert date/time to UTC from a defined time zone per project. 

Allow import of .Dux files for inclinometers

### **ASSUMPTIONS**
          1. Limited to CSV / TOA5 / DUX file formats
          2. Will be limited to single timestamp format (MM-DD-YYYY HH:MM)
          3. Time zones are defined at project level, and a datalogger can only be registered under one project.

dennisgsmith, 2024-10-04:
1. Will timezone of data uploaded be in UTC or local timezone or does this need to be configurable at upload?
2. Should the user-created equivalency mappings be available for all to use/edit within the project? Or private to users? What permissions are needed?
3.  Assumes we enforce standardized CSV / TOA5 / DUX formats

Rockscientist, 2024-10-04:
1. Configurable at upload, and remember the previous selection 2. Available to any admin user on the project. Once we have a mapping completed we don’t want to duplicate that effort for other users.ScottSent from my iPhone

dennisgsmith, 2024-11-21:
Closed with
https://github.com/USACE/instrumentation-api/releases/tag/v3.5.0
https://github.com/USACE/instrumentation-ui/releases/tag/v0.18.0

closed 2024-11-21 by dennisgsmith