USACE/instrumentation #183: MIDAS/OpenGround ArcPro Toolbar (2b)

madisonschattenfield, 2024-08-23:
### **DESCRIPTION**

Create an ArcPro toolbar that accesses the respective MIDAS and Openground databases.

1. Create “Import MIDAS” button, all instruments within the view window will be created as a feature class with a completed attribute table. Symbology shall be according to instrument type in the attribute table. Populate attribute table with relevant information from Projects( FederalID/SegmentID, Project Name, Division, District, etc) and Instruments (Instrument Name, Status, Type, etc) and a Hyperlink to the instrument in MIDAS displaying the Chart Screen. Schema/data table design will be provided.
2. Develop data connection with OpenGround using ESRI Koop or other method. Ensure code isnot available to the public, especially OpenGround Credentials. 
3. Create “Import OpenGround” button, Boreholes within the view window will be created as a point feature class with a completed attribute table. Schema/data table design will be provided.
4. Create “Import OpenGround 3D” button, Boreholes within the view window will be created as 3Dline feature class with a completed attribute table. Apply consistent symbology to the 3d lines. Schema/data table design will be provided.

### **ASSUMPTIONS**

1. Have defined set of relevant info from projects, ie defined fields that they can set
2. Ability to securely inject API credentials
3. OpenGround provides accessible API that will respond in compatible format in a timely manner

### **RISKS**

1. Keeping code hidden
2. Hide API Source

ScottMassa, 2024-08-30:
Some clarification.. just need to keep the OpenGround Account details and/or API key hidden.

jakefulkerson, 2024-12-03:
Began development of MIDAS import tool script in November. Hit an error with translating JSON from API into Feature Class in Pro.

Need to enable Bounding Box parameter for API call and experiment with MapBox Tile payload to see if that works better.

jakefulkerson, 2024-12-03:
Reviewed links provided by Chris for accessing OpenGround API and set up call for 12/4 at 12ET to talk with Scott Deaton (Data Forensics), Chris (USACE) and Ryan Benac (USACE) about previous work with accessing the API to get data into ArcGIS format.

jakefulkerson, 2024-12-16:
"""
Script documentation
- Tool parameters are accessed using arcpy.GetParameter() or 
                                     arcpy.GetParameterAsText()
- Update derived parameter values using arcpy.SetParameter() or
                                        arcpy.SetParameterAsText()
"""
import arcpy, requests, json5
def script_tool(param0):
    projdict = {}
    projurl = "https://develop-midas-api.rsgis.dev/projects"
    response = requests.get(projurl)
    data = response.json()
    for i in data:
        projdict[i["name"]] = i["id"]
    pid = projdict[param0]
    arcpy.AddMessage(pid)

    return
if __name__ == "__main__":
    param0 = arcpy.GetParameterAsText(0)
    script_tool(param0)

jakefulkerson, 2024-12-18:
Got OpenGround Cloud Launcher to work on GFE.

Having a work session with Dennis on Friday to work through ArcGIS SDK capabilities.

jakefulkerson, 2024-12-23:
We have MIDAS points being displayed in ArcGIS Pro, but attributes are not coming through and it is only the coordinates of the points. Might be an issue of not having "required" attributes from Esri.

jakefulkerson, 2024-12-23:
Vector Tile starting script:

"""
Script documentation

- Tool parameters are accessed using arcpy.GetParameter() or 
                                     arcpy.GetParameterAsText()
- Update derived parameter values using arcpy.SetParameter() or
                                        arcpy.SetParameterAsText()
"""
import arcpy
import requests
import os
import json

def script_tool():
    """Script code goes below"""
    queryUrl = "https://develop-midas-api.rsgis.dev/instruments/arcgis/vectortiles"
    r = requests.get(queryUrl)
    vt = r.json()
    return

if __name__ == "__main__":

    #param0 = arcpy.GetParameterAsText(0)
    #param1 = arcpy.GetParameterAsText(1)

    script_tool()
    arcpy.SetParameterAsText(2, "Result")

jakefulkerson, 2025-01-03:
"""
Script documentation

- Tool parameters are accessed using arcpy.GetParameter() or 
                                     arcpy.GetParameterAsText()
- Update derived parameter values using arcpy.SetParameter() or
                                        arcpy.SetParameterAsText()
"""
import arcpy
import json
import requests

import requests
import json

def pull_token():
    request_token_url = "https://imsoidc.bentley.com/connect/token"
    client_id = "XXXXX"
    client_secret = "XXXXX"

    body = {
        "grant_type": "client_credentials",
        "scope": "openground",
        "client_id": client_id,
        "client_secret": client_secret
    }

    token_response = requests.post(request_token_url, data=body)

    token = token_response.json()
    token = token['access_token']
    return token

def script_tool():
    """Script code goes below"""
    headers = {"Content-Type": "application/json"}
    getURL = 'https://api.eu-west-2-sandbox.OpenGround.cloud/api/v1.0/data/projects'
    r = requests.get(getURL, headers=headers)
    fc = r.json()
    arcpy.AddMessage(fc)
    return

if __name__ == "__main__":

    #param0 = arcpy.GetParameterAsText(0)
    #param1 = arcpy.GetParameterAsText(1)

    script_tool()
    arcpy.SetParameterAsText(2, "Result")

jakefulkerson, 2025-01-08:
New format for Python toolbox:

'''python
# -*- coding: utf-8 -*-

import arcpy
import requests
import os, sys
import json

class Toolbox:
    def __init__(self):
        """Define the toolbox (the name of the toolbox is the name of the
        .pyt file)."""
        self.label = "Toolbox"
        self.alias = "toolbox"

        # List of tool classes associated with this toolbox
        self.tools = [MIDASImport, MIDASVT, OpenGroundImport]

class MIDASImport:
    def __init__(self):
        """Define the tool (tool name is the name of the class)."""
        self.label = "MIDASImport"
        self.description = ""

    def getParameterInfo(self):
        """Define the tool parameters."""
        params = [
            arcpy.Parameter(displayName='Project',
                        name='project0',
                        datatype='GPString',
                        parameterType='Required',
                        direction='Input')
        ]
        
        params[0].filter.type = 'ValueList'
        params[0].filter.list = ['10th Street Pumping Station',
                              '17th Street Canal',
                              'Aberdeen',
                              'Abiquiu Dam',
                              'Blue Water Dam Example Project']
        return params

    def isLicensed(self):
        """Set whether the tool is licensed to execute."""
        return True

    def updateParameters(self, parameters):
        """Modify the values and properties of parameters before internal
        validation is performed.  This method is called whenever a parameter
        has been changed."""
        return

    def updateMessages(self, parameters):
        """Modify the messages created by internal validation for each tool
        parameter. This method is called after internal validation."""
        return

    def execute(self, parameters, messages):
        """The source code of the tool."""
        projectInput = parameters[0].valueAsText
        cwd = arcpy.env.scriptWorkspace
        arcpy.env.overwriteOutput = True
        # Get Extents
        aprx = arcpy.mp.ArcGISProject("CURRENT")
        mapView = aprx.listMaps()[0]
        new_sr = arcpy.SpatialReference("WGS 1984")
        mapView.spatialReference = new_sr
        aprx.save()
[... 190 lines cut ...]
            usace_inst_id = "570c1f2b-aed1-4791-a20b-4b8ac8cdd2c8"
            
            # specified by the API, in user documentation for OGC API
            return {
                'KeynetixCLoud': key_cloud, 
                'Authorization': f'Bearer {token}', 
                'Content-Type': 'application/json',
                'InstanceId': sand_inst_id,
                'scheme': 'https'
            }
        
        headers = {"Content-Type": "application/json"}
        getURL = 'https://api.eu-west-2-sandbox.OpenGround.cloud/api/v1.0/data/projects'
        r = requests.get(getURL, headers=headers)
        fc = r.json()
        arcpy.AddMessage(fc)
        return

        return

    def postExecute(self, parameters):
        """This method takes place after outputs are processed and
        added to the display."""
        return
'''

jakefulkerson, 2025-01-10:
OpenGround API - awaiting client_id and client_secret to retrieve authorization token as well as additional guidance from Ryan on best practices.

jakefulkerson, 2025-01-17:
Not using client_id, client_secret for authorizing data pulls. Tool is set up so that on first run, the user will be directed to a new window to login to Bentley and verify email. The user can copy and paste the authorization link from the browser to the input box and run the tool again and then it will retrieve the token. No sensitive information is stored in the script. User types in their email in the browser and the link used as the input is discarded after the script has finished running.

Script checks the current map view for extent and retrieves borehole data for projects that the user has access to and are within their view.

jakefulkerson, 2025-01-22:
MIDAS and OpenGround scripts are functioning and meet base requirements. OpenGround contacts are not familiar with 3D data available, so may need clarity that.

jakefulkerson, 2025-01-23:
Began development on 3D data tool. Units for Final Depth and Ground Level are available in a different API endpoint - haven't been able to get units yet.

jakefulkerson, 2025-01-23:
Alpha version of tool and documentation attached

[MIDAS_alpha.zip](https://github.com/user-attachments/files/18526789/MIDAS_alpha.zip)

madisonschattenfield, 2025-01-31:
Extended timeline to February Production Release to account for OpenGround 3D bore hole functionality.

jakefulkerson, 2025-02-14:
Set up troubleshoot meeting with Scott for this afternoon (2/14/2025).

Working on 3D data piece. Same process as 2D, but need units for Final Depth.

jakefulkerson, 2025-02-14:
Believe we found the issue for both tools.
- MIDAS tool was referencing the workspace where the Python toolbox was stored, rather than the project folder/Geodatabase.
- OpenGround was using the Sandbox instance ID rather than the USACE in the header for the requests.

Sent new zip file to clients for testing.

jakefulkerson, 2025-02-20:
1.	Resolved the issue noted in this screenshot of the MIDAS Import tool. Some projects have no information of the District they belong to, so now those data points will still be imported, but will have “No Associated District” for the District and “N/A” for the Division.
2.	Changed the name of the output feature class to be “import_<project slug>” because ArcGIS doesn’t like when a feature class name starts with an integer (10th Street, 17th Street, 28th).
3.	Adjusted the script so the json file that is used to create the feature class is stored in the project default folder. Also ensured that the script will retrieve the active map when running the tool.

I don’t have much of an update on the 3D import tool now, but we have a path forward for it. And we are also able to convert the 2D OpenGround import to a 3D feature class using the Final Depth field.

jakefulkerson, 2025-02-21:
Met with Scott Massa on 2/20 to discuss questions/status.
- Implementing "Import all within view" capability with limitations on scale.
- Add message in tool output to indicate how many points from the project(s) were pulled in based on extent.
- OG: Scott has 12,000+ projects in his USACE instance. Tool is set up to grab all projects and then check center point, but not feasible with that many projects. Need to see if there's a way to limit the number of projects checked based on map view extents (rather than getting all projects and checking center point).

jakefulkerson, 2025-03-21:
"Import all within view" option is configured and imports all points within the map view. Need to add a few more print statements:

- Total number of records in the map view to give an idea of how long the tool will take.
- Number of points per project imported compared to total number of project points available.

Will remove bounding box parameter from the project selection option.

Possible limitations:
- Only can run "Import all" at a certain scale threshold.
- Only select up to 5 projects in the selection window.