USACE/instrumentation #184: Seismic Data (3)

madisonschattenfield, 2024-08-23:
### **DESCRIPTION**

1. Create a new instrument type for seismic devices. The associated data will be uploaded by the Government. The seismic device data shall include the following data fields: ID, serial number, coordinates, make/model, installation date, maintenance date, operational status, and seismic readings associated with date/time. Readings shall include Peak Ground Accelerations (pga), Peak Ground Velocity, displacements, frequency readings over time and all associated data from USGS database required to produce graphical results. 
2. Establish Connection to ThingLogix API to collect state of health (SOH) data. SOH data includes battery voltage, temperature, etc. Retrieve data from USGS API using Airflow or other method every hour. Store data in appropriate timeseries. Show color coding symbol for operational status (green – online and functional, yellow warning, red- not-online). From the Seismic Monitoring screen, user should be able to select an instrument to get additional information and have an option to go to the associated local project (such as dam or levee) to see other instruments.
3. Establish Connection to ThingLogix API to collect Ground Accel Retrieve data from API using Airflow (or proposed alternative method) every hour. Store data in appropriate timeseries. 
4. Establish Connection to USGS Seedlink Server. This will require firewall actions and coordination with USGS. Automate collection of seed or miniseed files. Open files with already developed open-source software. Store in MIDAS Database. Display collected data in UI as a waveform and display associated data. 
5. Add notifications for seismic activities viewed within a dashboard, such as QA/QC. Notification should alert for when new data is received. Explorer view should highlight instrument location in red and remain red when new data is received. selection of instrument should display acceleration information.
6. Show graphical representation of data when selected (similar functionality as other instruments already in MIDAS). Graphical representation should be appropriate for the seismic instrumentation. Example plots are as shown on the right (see Figure 1).

### **ASSUMPTIONS**

1. Seismic devies can be optionally assigned to borings, just as other instrument types can.
2. Depends on Seismic Data (a)
3. Connection information is provided if it is not an open API.
4. Volume of data collected is similar to the MIDAS sensors.
5. Operational status and other required fields is coming from ThingLogix API
6. Seismic instrument dashboard will be implemented within a project. Instruments shown on dashboard are be limited to instruments associated with the project.
7. Assuming infrastructure set up in Seismic Data (b) is complete.
8. Point of contact will be provided to USGS SeedLink Server.
9. All of this is related to the Seismic Monitoring project and no other project.
10. Provide a sample seed or mini seed file.
11. Plotly and/or Apache ECharts can handle their graphic needs.

dennisgsmith, 2024-10-25:
foundry **thinglogix api**
- project has been dormant
- would like to stand up new api for read only data but not sure about timeline
- there is a public api, need to coordinate how api is used
	- it't not a straightforward rest call
- DOES NOT HAVE WAVEFORM DATA

state of health data for **seedlink server**
- python sdk - opspy
- eathscope org - repo of seismic data produced software that connects to seedlink services and archive data
- slarchive?
waveform data can be encoded in ascii
- note: can only short timeframe of data for SOH
- seedlink api
- DOES HAVE WAVEFORM DATA
- firewall access

ground motion values are published to a website that can be scraped
- data is publicly available
- earthquake.usgs.gov has api for shapemap product

madisonschattenfield, 2024-12-06:
Need access to Thinglogix API and firewall access to Seedlink server (IP addresses provided but have yet to hear back from USGS team on either of these resources. Dennis will send follow up to help facilitate communication.

madisonschattenfield, 2025-01-17:
Seedlink server access is now available and that portion is moving forward. Still waiting ThingLogix API access.

dennisgsmith, 2025-08-21:
Closing as completed now that ThingLogix integration has been implemented. This functionality will be available in the next MIDAS release.

closed 2025-08-21 by dennisgsmith