USACE/instrumentation #185: MIDAS Improvements (7b)

madisonschattenfield, 2024-08-23:
### **DESCRIPTION**

Collection Groups should allow the user to customize the order, and have that order be retained. When adding a field to the collection group, the instrument picklist should be searchable and alphabetized.

ScottMassa, 2024-08-23:
FYI, The collection groups order started changing around when data was submitted for a user today, first this has been seen. SAB went from top to bottom of the list. (images below if they make it into Github.)
***@***.***
***@***.***

dennisgsmith, 2024-08-23:
@ScottMassa Thanks for the info -- your images did not make it to GitHub, I believe you would need to use the browser for this. Either way, this feature should take care of that issue. We can review how we're prioritizing these features and make any adjustments as needed in next week's Sprint Planning.

dennisgsmith, 2024-11-21:
Closed with
https://github.com/USACE/instrumentation-api/releases/tag/v3.5.0
https://github.com/USACE/instrumentation-ui/releases/tag/v0.18.0

closed 2024-11-21 by dennisgsmith