USACE/instrumentation #192: [BUG REPORT]:Plot screen is very slow and unresponsive

ScottMassa, 2024-08-23:
## Describe the bug
Plot screen does not work as expected, it is very slow and becomes non-responsive.

## Steps To Reproduce
Go to plot screen, click on a plot to render.

## Expected behavior
Has performance like it did a week ago.
[image]

## Actual behavior
Plots are slow or do not load. page hangs.
[image]

## Screenshots

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser Microsoft Edge
 - Version 127

## Smartphone (please complete the following information):
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

## Additional context
Add any other context about the problem here.

dennisgsmith, 2024-08-23:
Hi @ScottMassa --

I'm testing this on my end, but unable to reproduce this issue. I noticed that in the first screenshot, I see only one month of data is being queried. In the second screenshot you sent, where you said the page hangs, the "Lifetime" parameter is selected.

Still I'm testing on this same dataset (InclinometerC1, Bluestone dam) now, and I'm unable to reproduce the issue. The response times for most traces are under 500ms, with the longest taking ~2 seconds.

There is definitely room for improvement when it comes to the performance of querying historical timeseries data, but I'm unable to replicate any regression.

I do, however, see a bug that maybe affecting the query:
[image]

The `after` and `before` parameters are extremely low, even though "lifetime" is selected. This is likely the reason that no data is showing on the plot as well. @ScottMassa Were there any specific actions you took in MIDAS before encountering this issue and are you able to replicate the issue consistently?

I'm going to update this issue to describe this bug -- we can discuss performance more generally in our next meeting and discuss avenues for improvements on that front.

ScottMassa, 2024-08-23:
It’s hard to capture. I was talking to a user earlier this week and the plots hung. I click some and they are super fast and responsive. I think it might be when we pull in really large datasets such as pool and tailwater. not the CWMS ones.

Hi @ScottMassa<Blockedhttps://github.com/ScottMassa> --

Screenshot.2024-08-23.at.2.25.56.PM.png (view on web)<Blockedhttps://github.com/user-attachments/assets/789827aa-5303-4855-a0c3-a7d349d95cf7>

The after and before parameters are extremely low, even though "lifetime" is selected. This is likely the reason that no data is showing on the plot as well. @ScottMassa<Blockedhttps://github.com/ScottMassa> Were there any specific actions you took in MIDAS before encountering this issue and are you able to replicate the issue consistently?

madisonschattenfield, 2024-12-02:
Currently unable to reproduce issue.

madisonschattenfield, 2024-12-13:
Unable to reproduce issue and will reopen if we can get clear steps to reproduce.