USACE/instrumentation #194: [BUG REPORT]: Plotting CWMS Data

ScottMassa, 2024-08-30:
## Describe the bug
CWMS Data can be observed in the CWMS Timeseries tab. This data is pulled from CDA.
When plotting, the script references the TS UUID and no data is retrieved to be plotted. 
It appears that we are requesting too large of a range when plotting for CDA to respond to, and get a 500 Internal server error
I would specify just a few days of data hoping that would go through. 

## Screenshots
[image]
[image]

## Desktop (please complete the following information):
 - OS: Windows 10
 - Browser MS Edge

ext about the problem here.

madisonschattenfield, 2024-11-22:
Need to coordinate with CWMS team to gather additional information on this bug and its reproducibility

dennisgsmith, 2024-12-10:
Looks like CWMS data API is now bubbling up the reason we were encountering this error:

https://cwms-data.usace.army.mil/cwms-data/timeseries?name=Sutton-Lake.Elev.Inst.15Minutes.0.Raw-GOES&office=LRD&begin=2020-08-29T00:00:00.000-04:00

```json
{
  "time-series": {
    "query-info": {
      "time-of-query": "2024-12-10T21:18:17Z",
      "process-query": "PT0.769S",
      "format-output": "PT0.000S",
      "requested-start-time": "2020-08-29T04:00:00Z",
      "requested-end-time": "2024-12-10T21:18:17Z",
      "requested-format": "JSON",
      "requested-office": "LRD",
      "requested-items": [
        {
          "name": "Sutton-Lake.Elev.Inst.15Minutes.0.Raw-GOES",
          "unit": "EN",
          "datum": "NATIVE"
        }
      ],
      "total-time-series-retrieved": 1,
      "unique-time-series-retrieved": 1,
      "total-values-retrieved": 0,
      "unique-values-retrieved": 0
    },
    "quality-codes": {
      "comment": "The following quality codes are used in this dataset",
      "codes": []
    },
    "time-series": [
      {
        "office": "LRD",
        "name": "Sutton-Lake.Elev.Inst.15Minutes.0.Raw-GOES",
        "alternate-names": [],
        "error": "Not enough memory for CONNECT BY operation"
      }
    ]
  }
}
```

Specifically this line: **"error": "Not enough memory for CONNECT BY operation"**

@KevinJJackson, can we could check for this error in the response and bubble it up to the end user? Maybe some dialogue like "the sampling rate is too frequent for a time frame of this size, please select select a time series with a larger sampling rate or reduce the time window".

@ScottMassa please let us know if this approach works and provide any additional input.

KevinJJackson, 2024-12-12:
> Looks like CWMS data API is now bubbling up the reason we were encountering this error:

I'm not seeing this running from the application:
[image]

I'll continue to check if they propagate anything, but I can toss a generic error for the time being to at least inform the user that something went wrong and to try to use a smaller data set, as that will likely be the cause of most errors.

closed 2024-12-12 by KevinJJackson