USACE/instrumentation #199: OpenDCS Vulnerability Remediation

dennisgsmith, 2024-09-20:
instrumentation-dcs uses old JAR archive that needs to be recompiled to patch 3 CRITICAL vulnerabilities.

Patching is non-trivial since only JAR archive of code exists in vcs (source code was not uploaded).

Current plan of action:
- [ ] Use Java disassembler to recreate source code
- [ ] Create Dockerfile to automatically from source in future deployments
- [ ] Push to environments and test

dennisgsmith, 2024-11-22:
Status: After using the installer and redeploying, it seems that a number of outdated test packages are included in this installation and include more vulnerabilities. Currently awaiting repository code scan results to identify problem dependencies/code. POAM extended timeline to end of December.

Will reach out to team Enterprise Java SME for recommendations on alternate build mechanisms. Updates will be documented in this issue.

madisonschattenfield, 2024-12-13:
Dennis to submit ticket to CWBI with updated task definition and will coordinate deployment of said ticket.

madisonschattenfield, 2025-01-03:
This was pushed right due to required infrastructure changes, but CWBI team prioritizing. Should be before POAM.

closed 2025-01-15 by dennisgsmith