USACE/instrumentation #210: [BUG REPORT]: CWMS Timeseries Listings

ScottMassa, 2024-11-22:
## Describe the bug
When viewing certain listings for timeseries, not all records are retrieved. 

## Steps To Reproduce
1. Go to CWMS Timeseries
2. Click on New CWMS Timesieres
3. Select office "SWF"
4. See projects only go through the letter "H"

## Expected behavior
CWMS Timesiere lists all the projects

## Actual behavior
The query has a page size of 5000, which is not enough to show all the listings. There are currently 10964, so a page size of say 15000 may be more appropriate.
https://cwms-data.usace.army.mil/cwms-data/catalog/TIMESERIES?office=SWF&page-size=5000](https://cwms-data.usace.army.mil/cwms-data/catalog/TIMESERIES?office=SWF&page-size=5000)

## Screenshots
List stops at letter H
[image]

## Desktop (please complete the following information):
 - OS: Windows 11
 - Browser Microsoft Edge
 - Version 131

dennisgsmith, 2024-11-22:
@KevinJJackson

Ref API Doc: https://cwms-data.usace.army.mil/cwms-data/swagger-ui.html 

The catalog endpoint has a `like` query parameter for searching timesieres. I think the best way to resolve this would be to add that query param if there's text in the query string and debounce requests.

For scrolling the param list, there's a `page` query param that takes a hash of the previous request's `next-page` param.

ScottMassa, 2025-02-25:
Looking at the CWMS data plotted, it looks like we only display 1 page (500 records) of data, can we pull more than one page from the CDA API?

dennisgsmith, 2025-02-25:
@ScottMassa Thanks for the bug report. After reviewing this, I'm going to spin it off into another new ticket, as it relates to the timeseries data, not the timeseries entities themselves.

closed 2025-02-25 by dennisgsmith