USACE/instrumentation #213: [BUG REPORT]: Uploader - High Priority

ScottMassa, 2024-12-05:
## Describe the bug
Uploader fails to post data to the database. 

## Steps To Reproduce
Use a file that has multiple instruments
I've tried several date formats, the ISO8601 returns a 201 created code with x-request ID of c5f04a7f-defb-4aee-b099-7ebb7231b478

dennisgsmith, 2024-12-05:
Thank you for submitting this bug report -- indeed, I found a regression where there was some bad logic in the field mapper. There is an open PR with a fix: https://github.com/USACE/instrumentation-api/pull/246

@KevinJJackson and I are working toward getting this fix out asap, will close this ticket once deployed.

dennisgsmith, 2024-12-06:
Fixed with https://github.com/USACE/instrumentation-api/releases/tag/v3.5.1

closed 2024-12-06 by dennisgsmith