USACE/instrumentation #217: [BUG REPORT]: Uploaded values not visible

ScottMassa, 2024-12-17:
## Describe the bug
Data appears to upload, but cannot view data

## Steps To Reproduce
Upload tab, upload attached Airtemp_2.csv
Timestamp: IS08601
instrument: V, enabled
Map F1 values to the appropriate timeseries
[image]

## Expected behavior
Data to be uploaded, 

## Actual behavior
Data not observed as a timeseries, 
[image]
but may be observed in the collection groups window (51 degrees a month ago)
[image]

## Screenshots

[Airtemp_2.csv](https://github.com/user-attachments/files/18170626/Airtemp_2.csv)

## Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Browser [e.g. Microsoft Edge]
 - Version [e.g. 90.0.818.46]

## Smartphone (please complete the following information):
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

## Additional context
Add any other context about the problem here.

ScottMassa, 2024-12-17:
Also on production, another user can see the timeseries table and I cannot see any data. 
[image]
vs 
[image]
With a 500 Internal server error

dennisgsmith, 2024-12-19:
Fix deployed: https://github.com/USACE/instrumentation-api/releases/tag/v3.5.2