USACE/instrumentation #229: [BUG REPORT]: Chart View is slow to repond with large data sets

ScottMassa, 2025-03-13:
## Describe the bug
The chart view pulls 1 year of data for the timeseries and formulas. This can impact the performance of the chart loading.

Consider changing the threshold from 0 to a default of say 2500 records, or a smaller default time span

## Expected behavior
Displays chart in a reasonable amount of time

## Actual behavior
charts displayed 35k records for 4 timesieries and took several minutes to load. 

## Screenshots
The image below took 350,000 ms or almost 6 minutes to populate
[image]

## Desktop (please complete the following information):
 - OS: Win11
 - Browser Microsoft Edge
 - Version 134

dennisgsmith, 2025-03-13:
@KevinJJackson Since the plot configuration is not saved at the project level, I think it would make the most sense to query a small amount of records here. Maybe based on the number of timeseries requested for a plot, it could be dynamically set by default?

For example
1 timeseries threshold=2500
2 timeseries threshold=1250
3 timeseries threshold=833
4 timeseries threshold=625
etc.

This would make performance more consistent, in terms of rendering. @ScottMassa do you think that would be appropriate for the instrument page overview?

Alternatively, maybe a manual threshold control could be added, with one of the threshold factors set as the default?

ScottMassa, 2025-03-13:
If you could do a manual threshold slider like on the plot screen that would be ideal. I don’t like the idea of downsampling too much without the user knowing or being able to control it. I wouldn’t want to go below 2 readings per day, or 730~750 a year.

Scott

This would make performance more consistent, in terms of rendering. @ScottMassa<BLOCKEDhttps://github.com/ScottMassa> do you think that would be appropriate for the instrument page overview?

@KevinJJackson<BLOCKEDhttps://github.com/KevinJJackson> Since the plot configuration is not saved at the project level, I think it would make the most sense to query a small amount of records here. Maybe based on the number of timeseries requested for a plot, it could be dynamically set by default?

This would make performance more consistent, in terms of rendering. @ScottMassa<BLOCKEDhttps://github.com/ScottMassa> do you think that would be appropriate for the instrument page overview?

dennisgsmith, 2025-03-13:
@ScottMassa In that case, I think setting a simple default of 750 per timeseries for the instrument chart page makes the most sense, then allowing users to adjust it as needed.