USACE/instrumentation #234: [BUG REPORT]: Telemetry data not displaying

ScottMassa, 2025-04-23:
## Describe the bug
Data is posted from a datalogger to MIDAS Telemetry API. Datalogger returns Model and Serial number indicating success, but no JSON is shown in the UI. Program was working last week, but I cycled the API key and it no longer displays data in MIDAS, 

## Steps To Reproduce
Send data from datalogger using template

## Expected behavior
Data is displayed in the preview

## Actual behavior
No data is displayed, 
Preview is 404 Not Found, X-Request-Id 048459c1-ca2e-49ef-9d4b-c83244075675

## Screenshots
If applicable, add screenshots to help explain your problem.

[image]

## Desktop (please complete the following information):
 - OS: Win 11
 - Browser  Microsoft Edge
 - Version 135

## Additional context
Datalogger program worked last week but was haveing trouble with another program i was working on. I cycled the API key, but no success, I then deleted the datalogger in MIDAS and added another one.  Checked status of another project and and it temporarily showed up under the "Blue Water Dam Example Project", "CR6-Bench1" data logger until a CRTL+R refresh.

dennisgsmith, 2025-05-13:
@ScottMassa moving to this thread for discussion of the datalogger preview / telemetry data not displaying after cycling the api key. I've attempted to replicate this behavior with the mock datalogger Python script without success.

If you are able to, could you have the datalogger running against the Telemetry API so @KevinJJackson and I can look into what may be causing this?

dennisgsmith, 2025-05-15:
Fixed with https://github.com/USACE/instrumentation-api/commit/064b4a4afbab1d0ec30732679e11f59d05f3a84b

closed 2025-05-15 by dennisgsmith