USACE/instrumentation #251: Create new instrument parameters in Site Administration menu

dennisgsmith, 2025-07-11:
a. Create an interface that will allow site administrators to create new “Instrument
Type". New instruments types will default to the basic instrument page (i.e., it will not have unique instrument views).
b. Create an interface that will allow site administrators to create new “parameters”
and “units” for a timeseries.
c. Add "Distress indicators" instrument attribute

ScottMassa, 2025-08-26:
I think we would want a distress indicator icon, then be able to classify it as a subcategory such as: sinkhole, wet area, sand boil, slide, erosion, etc.

dennisgsmith, 2025-08-26:
@ScottMassa To clarify, do you mean we need another domain, "Distress Indicators" (name, icon), that can be managed by application admins? Then one (or many?) available "Distress Indicators" could be assigned to an instrument by a Project Admin?