USACE/instrumentation #254: [BUG REPORT]: Plotting of Calculated Timeseries

ScottMassa, 2025-07-16:
## Describe the bug
Calculated Timeseries fail to plot on 5 year scale. Inspector shows multiple points at date 0001-01-01T00:00:00

## Steps To Reproduce
1. Go to Paint Creek Dam
2. Click on Plotting, "Left Abutment"
3. View 1 year plot, see that it works, view 5 year plot and no dashed lines appear.

## Expected behavior
both 1 year and 5 year plots should show dashed lines

## Actual behavior
5 year plot did not display calculated timeseries

## Screenshots
If applicable, add screenshots to help explain your problem.

[image]

## Desktop (please complete the following information):
 - OS: Win 11
 - Browser Microsoft Edge
 - Version 138

dennisgsmith, 2025-09-19:
@ScottMassa I believe I've identified the cause of this issue; the dependency timeseries here has a measurement in the database corresponding to time 0001-01-01T00:00:00Z and value NaN. After removing this value, the plot seems to work as expected.

I've improved our In our LOCF implementation; we were carrying forward this value, causing each point to be overwritten with this zero time and value. I've created a PR that should resolve this in future cases. This will be deployed in the next release.

dennisgsmith, 2025-09-26:
Closed with https://github.com/USACE/instrumentation-api/pull/282

closed 2025-09-26 by dennisgsmith